riscv-software-src/riscv-pk #236: Error during the building

vulkomilev, 2021-04-01:
When I try to build the project I get this `../configure --prefix=$RISCV --host=riscv64-unknown-elf
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-elf
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... no
configure: error: in `/home/<username>/RISC-V/riscv-pk/build':
configure: error: C compiler cannot create executables
See `config.log' for more details
`
here is the dump from the log file
`This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by RISC-V Proxy Kernel configure 1.0.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ../configure --prefix=/home/<username>/RISC-V/riscv-gnu-toolchain/ --host=riscv64-unknown-elf

## --------- ##
## Platform. ##
## --------- ##

hostname = <username>-System-Product-Name
uname -m = x86_64
uname -r = 5.8.0-45-generic
uname -s = Linux
uname -v = #51~20.04.1-Ubuntu SMP Tue Feb 23 13:46:31 UTC 2021

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /opt/openfoam8/bin
PATH: /opt/openfoam8/platforms/linux64GccDPInt32Opt/bin
PATH: /home/<username>/anaconda3/bin
PATH: /home/<username>/anaconda3/condabin
PATH: /home/<username>/.local/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /usr/games
PATH: /usr/local/games
PATH: /snap/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:1981: checking build system type
configure:1995: result: x86_64-pc-linux-gnu
configure:2015: checking host system type
configure:2028: result: riscv64-unknown-elf
configure:2061: checking for riscv64-unknown-elf-gcc
configure:2077: found /usr/bin/riscv64-unknown-elf-gcc
configure:2088: result: riscv64-unknown-elf-gcc
configure:2357: checking for C compiler version
configure:2366: riscv64-unknown-elf-gcc --version >&5
riscv64-unknown-elf-gcc () 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
[... 165 lines cut ...]
prefix='/home/<username>/RISC-V/riscv-gnu-toolchain'
program_transform_name='s,x,x,'
psdir='${docdir}'
runstatedir='${localstatedir}/run'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
stow=''
subprojects=''
subprojects_enabled=''
sysconfdir='${prefix}/etc'
target_alias=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "RISC-V Proxy Kernel"
#define PACKAGE_TARNAME "riscv-pk"
#define PACKAGE_VERSION "1.0.0"
#define PACKAGE_STRING "RISC-V Proxy Kernel 1.0.0"
#define PACKAGE_BUGREPORT "Andrew Waterman"
#define PACKAGE_URL ""

configure: exit 77`

vulkomilev, 2021-04-01:
The os is ubuntu 20.04

Wlfc912, 2023-02-20:
@vulkomilev I'm now having the same problem. Have you found a solution to this?