riscv-software-src/riscv-pk #238: mmap.c::__page_alloc() calls memset

zinechant, 2021-04-08:
This would pollute caches and produce strange cache numbers in spike-isa-sim. Especially for the case where compute is not intensive.
Opening this issue to open the discussion on whether this memset is necessary.

aswaterman, 2021-04-08:
It is necessary.  mmap'ed pages are required to be zeroed, and of course there's no guarantee that main memory is already zeroed.  (The fact that Spike zeroes main memory at boot is an implementation detail and is not true of other RISC-V implementations.)

Besides mmap, other uses of __page_alloc within pk rely on the zero initialization, including when new page tables are allocated.

aswaterman, 2021-04-08:
Search for the word "zero-fills" on this page for more details on the mmap situation: https://pubs.opengroup.org/onlinepubs/7908799/xsh/mmap.html

zinechant, 2021-04-08:
That was fast! Thanks!
I am happy with a little hack that commenting out the memset for my purpose.
Going to close this issue.

closed 2021-04-08 by zinechant

aswaterman, 2021-04-08:
Cool, good luck.