riscv-software-src/riscv-pk #239: Calls to __clear_cache() require pointer arguments

cmuellner, 2021-04-28:
When compiling with a recent version of GCC, we run into the following error:

```
      ../pk/pk.c: In function 'run_loaded_program.constprop':
      ../pk/pk.c:177:3: error: both arguments to '__builtin___clear_cache'
      must be pointers
        177 |   __clear_cache(0, 0);
            |   ^~~~~~~~~~~~~~~~~~~
```

The error comes from the fact, that we pass numeric values as arguments to the invocation of [__clear_cache](https://github.com/riscv/riscv-pk/blob/master/pk/pk.c#L177).

Using `(void*)0` or `NULL` instead of `0` does not work either.

My understanding of the code is, that we know that we can only clear the complete instruction cache and therefore we don't care about the arguments. Is this assumption correct?

This error has been introduced by the GCC commit [c05ece9 in Dec 2020](https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=c05ece92c6153289fd6055e31e791e59b8ac412).
I will file this in the GCC bugzilla as well.

cmuellner, 2021-04-28:
Ping for @aswaterman as he is the author of the affected piece of code.

aswaterman, 2021-04-28:
I think we can replace this one with an asm volatile fence.i.  If you make such a PR, I’ll merge it.

cmuellner, 2021-04-28:
The required code was already there. Just needed to be called.
Thanks!

aswaterman, 2021-04-29:
Resolved by #240.

closed 2021-04-29 by aswaterman