riscv-software-src/riscv-pk #259: Accessing user space data from machine mode.

msingla403, 2022-01-15:
I am trying to print a string from machine mode. The string is stored in user mode space.

### In user space:

buff : "Hello World\n"
&buff (virtual address in user space)= `0x000000000001f480`

### In machine mode:

addr = Physical address of buff = `0x000000408001f480`
`printm("%s"\n", addr);` leads to **exception trap_load_access_fault**

### PMP entries:
```
csrw    pmpaddr0, -1
csrw    pmpcfg0, 0x1f
```
By default in `PK`, PMP are set to permit access to all of memory.

Does anyone know of the issue? To access user space from Supervisor mode, we need to set `SUM` bit in `mstatus`. Are there any similar permission settings that I need to set? Or is there any another way to do this?

Thanks in advance

aswaterman, 2022-01-16:
There's nothing like SUM needed in this case.  My best guess is that the physical address is incorrect. (`0x408001f480` is very large; it suggests you've configured Spike to use more than 512 GiB of memory.)

msingla403, 2022-01-16:
Yes, the physical address is wrong. I used `kva2pa` to convert the user-space virtual address to the physical address. As the kernel and user-space are present in the same virtual address space, I assumed `kva2pa` should work. 

I will have to use `walk` on page tables to get the correct physical address. Is there any other way? `kva2pa` works for kernel-space virtual address as the kernel is present at constant offset.

aswaterman, 2022-01-16:
`kva2pa` only works on kernel virtual addresses.  `walk` is the way to go.

msingla403, 2022-01-20:
I am able to access user space contents. Used `walk` to retrieve physical address.

closed 2022-01-20 by msingla403