riscv-software-src/riscv-pk #260: Error: unrecognized opcode `fence.i'

srsorry, 2022-01-21:
Hello.
When I build the rocket-tools prj by ./build, it return error with unrecognized opcode `fence.i' on rocket.pk.
The error information below：
Removing existing riscv-pk/build directory
Configuring project riscv-pk
Building project riscv-pk
../dummy_payload/dummy_entry.S: Assembler messages:
../dummy_payload/dummy_entry.S:23: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:24: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:25: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:26: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:27: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:28: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:29: Warning: unterminated string; newline inserted
../machine/mentry.S: Assembler messages:
../machine/mentry.S:93: Error: unrecognized opcode `fence.i'
gmake: *** [mentry.o] Error 1
gmake: *** Waiting for unfinished jobs....
../machine/flush_icache.c: Assembler messages:
../machine/flush_icache.c:4: Error: unrecognized opcode `fence.i'
gmake: *** [flush_icache.o] Error 1

aswaterman, 2022-01-21:
Try configuring with `--with-arch=rv64gc_zifencei` (or whatever is appropriate for your target).  More recent assemblers require the `zifencei` extension to be explicitly listed for the `fence.i` instruction to be available.

srsorry, 2022-01-21:
THX
I use   CC= CXX= build_project riscv-pk --prefix=$RISCV --host=riscv64-unknown-elf --with-arch=rv64gc_zifencei now.
But it seems not work.
The new error shows below:
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'

The full error :
Configuring project riscv-pk
Building project riscv-pk
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
gmake: *** [file.o] Error 1
gmake: *** Waiting for unfinished jobs....
gmake: *** [handlers.o] Error 1
gmake: *** [console.o] Error 1
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
gmake: *** [elf.o] Error 1
gmake: *** [logo.o] Error 1
gmake: *** [dummy_payload.o] Error 1
gmake: *** [entry.o] Error 1
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
cc1: error: -march=rv64gc_zifencei: unsupported ISA substring '_zifencei'
gmake: *** [syscall.o] Error 1
gmake: *** [frontend.o] Error 1
gmake: *** [usermem.o] Error 1
gmake: *** [mmap.o] Error 1
gmake: *** [f32_lt.o] Error 1
gmake: *** [f32_add.o] Error 1
gmake: *** [f32_eq.o] Error 1
gmake: *** [f32_div.o] Error 1
gmake: *** [f32_mul.o] Error 1
gmake: *** [dummy_entry.o] Error 1
gmake: *** [f32_lt_quiet.o] Error 1
gmake: *** [f32_sqrt.o] Error 1

srsorry, 2022-01-21:
Maybe I need use other ISA.

aswaterman, 2022-01-21:
It seems you are having some problems with your compiler. I’d help if I could, but since I can’t replicate your environment, it’s not practIcal for me to help.

While I understand your frustration, suggesting you’ll use another ISA isn’t going to get you better support. I encourage you to use whatever ISA best suits your needs.

srsorry, 2022-01-24:
After I re-downloaded the whole project of rocket-tools and recompiled, the problem went away. So I'm also not sure where the problem is.
Anyway, thanks a lot for your help.

gustad, 2023-09-24:
I just cloned `ca6dc52` and did a `./build.sh` which resulted in 
```
../machine/flush_icache.c: Assembler messages:
../machine/flush_icache.c:4: Error: unrecognized opcode `fence.i', extension `zifencei' required
gmake: *** [Makefile:336: flush_icache.o] Error 1
```

Shreesh-Kulkarni, 2023-11-11:
> Try configuring with `--with-arch=rv64gc_zifencei` (or whatever is appropriate for your target). More recent assemblers require the `zifencei` extension to be explicitly listed for the `fence.i` instruction to be available.

This again leads to another error regarding the `stubs-lp64.h` file from the GNU compiler not being detected. This happens only when specific architectures configurations are used such as the `--with arch` as you said. 
However, this issue is already solved, (although seems like a informal one) here : [Issue:125](https://github.com/riscv-software-src/riscv-pk/issues/125). Just putting it out there for people who get this error.

TT430, 2023-11-20:
Recompiling riscv-gnu-toolchain with --with-arch=rv64gc_zfencei maybe help. This issue seems due to that your assembler doesn't support instruction 'fence.i'

Vico304, 2024-07-23:
I've had similar problems in other situations as well, and in addition to using the flag "--with-arch=rv64gc_zfencei" when compiling, I ended up also including the parameter "PLATFORM_RISCV_ISA=rv64imafdc_zifencei" in make operation, and that fixed the problem