riscv-software-src/riscv-pk #267: bbl: change-address not recognized by llvm-objcopy

ZenithalHourlyRate, 2022-02-06:
In `bbl/bbl.mk.in`, bbl.bin is built using change-addresses.
```
$(OBJCOPY) -S -O binary --change-addresses -@MEM_START@ $< $@
```
However, this is not recognized by `llvm-objcopy`.
```
llvm-objcopy: error: unknown argument '--change-addresses'
```
Instead, GNU binutils `objcopy` and `llvm-objcopy` both have an option called `--change-start`, but I do not know whether this option works as I have not tried bbl under my workflow (my workflow only needs `pk` thus I have to invoke `make pk` instead of just`make`)

aswaterman, 2022-02-07:
I don't have any way of testing this, either.  Until someone can verify a fix for both, assume that you must use GNU objcopy for this.

MaskRay, 2022-02-09:
As an llvm-objcopy contributor: llvm-objcopy does not support --change-addresses/--change-section-address (https://github.com/llvm/llvm-project/issues/44562#issuecomment-981025564). Just changing `sh_addr` is not difficult to implement, but the option seems to imply some layout changes which are poorly specified. The poor specification and our unawareness of how users (riscv-pk is one, but others may use this differently) use this option makes it somewhat infeasible to implement in llvm-objcopy. I implemented --only-keep-debug by (nearly) fully understanding its behavior. For --change-addresses/--change-section-address, I'd expect something similar.

If your demand can be achieved with a linker script, you may consider a linker script command.

subnut, 2024-12-21:
https://github.com/llvm/llvm-project/pull/98664