riscv-software-src/riscv-pk #269: Spike pk doesn't work

hasver, 2022-02-21:
Hi all 
I installed pk and spike both in my RISCV_TOOLCHAIN path and according to the instructions provided. 
I tried to run `spike pk hello` on an arbitrary c file but it just stops and doesn't show anything for a while 
Is there any solution for that? 
Thanks

tak-ka3, 2022-11-27:
Hi, @hasver 
I have the same problem like this(https://github.com/riscv-software-src/riscv-isa-sim/issues/1152). How did you resolve it?

mhamdan91, 2023-02-23:
It seems pk falls in bad_trap! running into the same problem

adamsir32, 2023-04-07:
I have the same question,that's terrible:(

kaichsu, 2024-07-08:
I've run into the problem, the situation is weird.
I compile and install pk like 
```
../configure --prefix=$RISCV --host=riscv64-unknown-elf --with-arch=rv64gc_zifencei
make
make install
```

And I write a simple hello world program
```
#include <stdio.h>
int main(){
    printf("hello riscv!\n");
    return 0;
}
```

Compile the `hello.c` program with the following command and run
```
riscv64-unknown-elf-gcc -g -march=rv64gc_zifencei hello.c -o hello
spike --isa=rv64gc pk hello
```

The terminal show the following and then stuck
```
bbl loader
```

If I run debug mode with `spike -d --isa=rv64gc pk hello` and let it run
The terminal eventually shows exception and stuck.
```
core   0: exception trap_instruction_page_fault, epc 0x00000000800019d0
core   0:           tval 0x00000000800019d0
```

However, if I remove the newline `\n` in `printf()` function. Recompile and run.
```
bbl loader
hello riscv!{my user name}:/tmp$
```
I have no idea why newline matters.