riscv-software-src/riscv-pk #270: Seeking Help.

WajahatRiaz, 2022-02-22:
I'm trying to install proxy kernel for spike (RISC V software).
Firstly, I used 
`https://github.com/riscv-software-src/riscv-pk.git pk-demo`
followed by
`../configure --prefix=$/home/whitebeard/riscv-demo --host=riscv64-unknown-elf`
then
`sudo make install`

The error I get is as follows: (screenshot attached for your perusal)
[image]

aswaterman, 2022-02-22:
See #204

closed 2022-02-22 by aswaterman

WajahatRiaz, 2022-02-22:
> See #204 
Yes I looked into it. Still didn't solve anything. Perhaps I'm making a mistake. Can you elaborate how the problem solved for you.

saad-waheed, 2022-02-25:
@WajahatRiaz can you please send the output log when you execute the `.\configure ...` command?

WajahatRiaz, 2022-02-25:
@saad-waheed Well the git repository I was using had bugs. I cloned the individual packages sperately in a folder and built them. It solved the issue. You have to add the path in the environment as well using the export command.

saad-waheed, 2022-02-26:
Great 👍