riscv-software-src/riscv-pk #275: Errors when running build steps

chiachia36, 2022-04-06:
Hi,
Here is my comment before `make`.
`../configure --prefix=$RISCV_TOOLCHAIN_ROOT_32 --host=riscv32-unknown-elf --with-arch=rv32i`
Here is error message I got.
```
../pk/entry.S: Assembler messages:
../pk/entry.S:57: Error: unrecognized opcode `csrrw sp,sscratch,sp'
../pk/entry.S:59: Error: unrecognized opcode `csrr sp,sscratch'
../pk/entry.S:61: Error: unrecognized opcode `csrrw t0,sscratch,x0'
../pk/entry.S:61: Error: unrecognized opcode `csrr s0,sstatus'
../pk/entry.S:61: Error: unrecognized opcode `csrr t1,sepc'
../pk/entry.S:61: Error: unrecognized opcode `csrr t2,stval'
../pk/entry.S:61: Error: unrecognized opcode `csrr t3,scause'
../pk/entry.S:70: Error: unrecognized opcode `csrw sscratch,sp'
../pk/entry.S:76: Error: unrecognized opcode `csrw sstatus,t0'
../pk/entry.S:77: Error: unrecognized opcode `csrw sepc,t1'
Makefile:336: recipe for target 'entry.o' failed
make: *** [entry.o] Error 1
make: *** Waiting for unfinished jobs....
```
Here is tool list in $RISCV_TOOLCHAIN_ROOT_32/bin.
```
riscv32-unknown-elf-addr2line   riscv32-unknown-elf-gcc-nm         riscv32-unknown-elf-nm
riscv32-unknown-elf-ar          riscv32-unknown-elf-gcc-ranlib     riscv32-unknown-elf-objcopy
riscv32-unknown-elf-as          riscv32-unknown-elf-gcov           riscv32-unknown-elf-objdump
riscv32-unknown-elf-c++         riscv32-unknown-elf-gcov-dump      riscv32-unknown-elf-ranlib
riscv32-unknown-elf-c++filt     riscv32-unknown-elf-gcov-tool      riscv32-unknown-elf-readelf
riscv32-unknown-elf-cpp         riscv32-unknown-elf-gdb            riscv32-unknown-elf-run
riscv32-unknown-elf-elfedit     riscv32-unknown-elf-gdb-add-index  riscv32-unknown-elf-size
riscv32-unknown-elf-g++         riscv32-unknown-elf-gprof          riscv32-unknown-elf-strings
riscv32-unknown-elf-gcc         riscv32-unknown-elf-ld             riscv32-unknown-elf-strip
riscv32-unknown-elf-gcc-11.1.0  riscv32-unknown-elf-ld.bfd
riscv32-unknown-elf-gcc-ar      riscv32-unknown-elf-lto-dump
```

Did I miss any steps?
Thanls!

aswaterman, 2022-04-06:
I'm just shooting in the dark here, but you might need to add `$RISCV_TOOLCHAIN_ROOT_32/bin` to your PATH before executing `configure`.  If `configure` can't locate the RISC-V compiler, it'll end up using your host compiler.

canoriz, 2022-04-12:
It seems the problem is mentioned at [https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1053](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1053). I revert my ```riscv-gnu-toolchain``` to tag:2022.03.09.