riscv-software-src/riscv-pk #283: exception trap_instruction_address_misaligned, epc 0x8000029c

ImanHosseini, 2022-09-14:
I have a simple program compiled with:
```
riscv64-unknown-elf-gcc -mabi=ilp32 -nostdlib -march=rv32im -o t2 t2.c
```
and I try to run it with spike as:
```
spike --isa=rv32im -d pk t2
```
And it crashes with "trap_instruction_address_misaligned", here's a trace:
```
core   0: 0x00001000 (0x00000297) auipc   t0, 0x0
: 
core   0: 0x00001004 (0x02028593) addi    a1, t0, 32
: 
core   0: 0x00001008 (0xf1402573) csrr    a0, mhartid
: 
core   0: 0x0000100c (0x0182a283) lw      t0, 24(t0)
: 
core   0: 0x00001010 (0x00028067) jr      t0
: 
core   0: 0x80000000 (0x1f80006f) j       pc + 0x1f8
: 
core   0: 0x800001f8 (0x00000093) li      ra, 0
: 
core   0: 0x800001fc (0x00000113) li      sp, 0
: 
core   0: 0x80000200 (0x00000193) li      gp, 0
: 
core   0: 0x80000204 (0x00000213) li      tp, 0
: 
core   0: 0x80000208 (0x00000293) li      t0, 0
: 
core   0: 0x8000020c (0x00000313) li      t1, 0
: 
core   0: 0x80000210 (0x00000393) li      t2, 0
: 
core   0: 0x80000214 (0x00000413) li      s0, 0
: 
core   0: 0x80000218 (0x00000493) li      s1, 0
: 
core   0: 0x8000021c (0x00000613) li      a2, 0
: 
core   0: 0x80000220 (0x00000693) li      a3, 0
: 
core   0: 0x80000224 (0x00000713) li      a4, 0
: 
core   0: 0x80000228 (0x00000793) li      a5, 0
: 
core   0: 0x8000022c (0x00000813) li      a6, 0
: 
core   0: 0x80000230 (0x00000893) li      a7, 0
: 
core   0: 0x80000234 (0x00000913) li      s2, 0
: 
core   0: 0x80000238 (0x00000993) li      s3, 0
: 
core   0: 0x8000023c (0x00000a13) li      s4, 0
: 
core   0: 0x80000240 (0x00000a93) li      s5, 0
: 
core   0: 0x80000244 (0x00000b13) li      s6, 0
: 
core   0: 0x80000248 (0x00000b93) li      s7, 0
: 
core   0: 0x8000024c (0x00000c13) li      s8, 0
: 
core   0: 0x80000250 (0x00000c93) li      s9, 0
: 
core   0: 0x80000254 (0x00000d13) li      s10, 0
: 
core   0: 0x80000258 (0x00000d93) li      s11, 0
: 
core   0: 0x8000025c (0x00000e13) li      t3, 0
: 
core   0: 0x80000260 (0x00000e93) li      t4, 0
: 
core   0: 0x80000264 (0x00000f13) li      t5, 0
: 
core   0: 0x80000268 (0x00000f93) li      t6, 0
: 
core   0: 0x8000026c (0x34001073) csrw    mscratch, zero
: 
core   0: 0x80000270 (0x00000297) auipc   t0, 0x0
: 
core   0: 0x80000274 (0xd9428293) addi    t0, t0, -620
: 
core   0: 0x80000278 (0x30529073) csrw    mtvec, t0
: 
core   0: 0x8000027c (0x30502373) csrr    t1, mtvec
: 
core   0: 0x80000280 (0x00629063) bne     t0, t1, pc + 0
: 
core   0: 0x80000284 (0x00010117) auipc   sp, 0x10
: 
core   0: 0x80000288 (0xc3c10113) addi    sp, sp, -964
: 
core   0: 0x8000028c (0xf14026f3) csrr    a3, mhartid
: 
core   0: 0x80000290 (0x00c69613) slli    a2, a3, 12
: 
core   0: 0x80000294 (0x00c10133) add     sp, sp, a2
: 
core   0: 0x80000298 (0x00069463) bnez    a3, pc + 8
: 
core   0: 0x8000029c (0x2e60306f) j       pc + 0x32e6
core   0: exception trap_instruction_address_misaligned, epc 0x8000029c
core   0:           tval 0x80003582
```

aswaterman, 2022-09-14:
`pk` was compiled using the C extension, but you are running Spike without the C extension.

ImanHosseini, 2022-09-14:
Thanks! I see... so that address is unaligned because compressed instructions can be aligned at 2-byte boundaries instead of 4 right? (so the jump target is a C- instruction) 
So I now tried this (added f too because it was using C.fswsp):
```
spike --isa=rv32imfc -d pk t2
```
Now it traps a few instructions later at:
```
core   0: 0x8000029c (0x2e60306f) j       pc + 0x32e6
: 
core   0: 0x80003582 (0x00007179) c.addi16sp sp, -48
: 
core   0: 0x80003584 (0x0000f406) c.fswsp ft1, 40(sp)
core   0: exception trap_illegal_instruction, epc 0x80003584
core   0:           tval 0x0000f406
```
Which is odd, C.FSWSP should be covered with *fc ?

aswaterman, 2022-09-14:
> so that address is unaligned because compressed instructions can be aligned at 2-byte boundaries instead of 4 right?

Yeah, that's right.

> C.FSWSP should be covered with *fc ?

Yeah, but it's possible the FPU hasn't been enabled yet (by setting mstatus.FS).  Unfortunately, I can't help dig into debugging that.

ImanHosseini, 2022-09-14:
Thanks! I will try to take it from here :) Just 1 question: all of this is _not_ the code in my program right? Based on the addresses, these are all the 'pk' startup code (my code has no C- instructions at all), so is this a bug on pk side?

aswaterman, 2022-09-14:
Right, the addresses being >= 0x80000000 suggest to me it's part of `pk`.

closed 2022-09-14 by ImanHosseini

aswaterman, 2022-09-14:
Oh, nice!

ImanHosseini, 2022-09-15:
Unlike cases of unsupported extension: where the error message specifically tells you what went wrong, cases of RV32/RV64 incompatibilities like this lead to unhelpful errors and you shouldn't have loaded an incompatible code in the first place - & if you have, it means you are not aware that you are doing it, like here - 
I think it would be nice if spike didn't load incompatible code and just told you that you are making that mistake. PR: https://github.com/riscv-software-src/riscv-isa-sim/pull/1085