riscv-software-src/riscv-pk #286: error in ./build-setup.sh riscv-tools process

yjf111, 2022-11-28:
in `chipyard` project Repository set up , i met error in the process of `./build-setup.sh riscv-tools`   
these are the **error message:**
```
...
make: Leaving directory '/home/yjf/miniconda3/chipyard/toolchains/riscv-tools/riscv-isa-sim/build'
==>  Installing Proxy Kernel
=>  Starting riscv-pk build
==>  Initializing riscv-pk submodule
+ git submodule update --init --recursive /home/yjf/miniconda3/chipyard/toolchains/riscv-tools/riscv-pk
error: Server does not allow request for unadvertised object e8e6b3aaee44d43b48164fbd377864c3a682dbd3
fatal: Fetched in submodule path 'toolchains/riscv-tools/riscv-pk', but it did not contain e8e6b3aaee44d43b48164fbd377864c3a682dbd3. Direct fetching of that commit failed.
```
 i can't find any solutions for this by search engine. How do i solve this problem? THANKS for suggestions :)

aswaterman, 2022-11-28:
This looks like a build system specific to chipyard.  You should ask there instead.

yjf111, 2022-11-28:
> 
OK, thanks!