riscv-software-src/riscv-pk #297: spike pk hello, failed with assertion failed: !(eh.e_flags & EF_RISCV_RVC)

jundijiujieke, 2023-05-03:
spike pk hello, failed with assertion failed: !(eh.e_flags & EF_RISCV_RVC)
Does any know how to fix this probelm? I don't know whether is my riscv-tool's problem or pk's problem?

bbl loader
../pk/elf.c:47: assertion failed: !(eh.e_flags & EF_RISCV_RVC)
Power off

aswaterman, 2023-05-03:
Looks like `pk` was compiled without support for the C extension, but you're trying to run a binary that requires the C extension.

jundijiujieke, 2023-05-16:
> Looks like `pk` was compiled without support for the C extension, but you're trying to run a binary that requires the C extension.

Thanks for your reply, do you mean i have to reinstall the pk with C extension.

aswaterman, 2023-05-16:
Yeah, I think building `pk` with support for the C extension will fix this issue.

jundijiujieke, 2023-05-17:
Thanks for your help, I tried your solution and fixed this problem.

aswaterman, 2023-05-17:
Great! Glad I was able to help.

closed 2023-05-17 by aswaterman