riscv-software-src/riscv-pk #298: [question] Kernel load segfault

WojciechMula, 2023-05-10:
I compiled the whole build toolchain from scratch, likewise `spike` and `pk`. When trying to run the simplest "Hello, World" program on my Linux box, I'm getting the following error: "Kernel load segfault @ 0x0000000010000005".

Any advice on what to check/recompile/whatever would be highly appreciated.

zacky1972, 2023-05-14:
I am also facing to this issue.

Distribution: Ubuntu 22.04

I built and installed `riscv-gnu-toolchain` as follows:

```bash
sudo apt update
git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git
cd riscv-gnu-toolchain
mkdir build
./configure --prefix=/usr/local/riscv
make -j$(nproc)
make install
```

I built and installed `spike` as follows:

```bash
sudo apt update
sudo apt install device-tree-compiler
git clone https://github.com/riscv-software-src/riscv-isa-sim.git
cd riscv-isa-sim
mkdir build
cd build
../configure --prefix=/usr/local/riscv
make -j$(nproc)
sudo make install
```

I built and installed `pk` as follows:

```bash
git clone https://github.com/riscv-software-src/riscv-pk.git
cd riscv-pk
mkdir build
cd build
../configure --prefix=/usr/local/riscv --host=riscv64-unknown-elf
make -j$(nproc)
sudo make install
```

I built an sample program:

```c
int main()
{
    return 0;
}
```

Then, compiled and executed it as follows:

```bash
/usr/local/riscv/bin/riscv64-unknown-elf-gcc -static hello.c 
/usr/local/riscv/bin/spike /usr/local/riscv/riscv64-unknown-elf/bin/pk a.out
```

Then, got the following error:

```
bbl loader
z  0000000000000000 ra ffffffc000004214 sp ffffffc00041ed30 gp 0000000000000000
tp 0000000000000000 t0 0000000000000025 t1 ffffffc00041ed6f t2 0000000000000000
s0 ffffffc00041ed31 s1 ffffffc00001a0c8 a0 000000000000002e a1 0000000000000100
a2 ffffffc00000d2b4 a3 0000000010000000 a4 0000000010000005 a5 0000000000000000
a6 000000000000003f a7 000000000000003f s2 ffffffc00001a0d8 s3 ffffffc00001a0e0
s4 ffffffc00000f018 s5 0000000000000001 s6 ffffffc000010478 s7 ffffffc00001a040
s8 ffffffc000010480 s9 ffffffc000010488 sA 0000000000000001 sB 0000000000000000
t3 fffffffffffffffc t4 ffffffc00000d060 t5 0000000000000009 t6 0000000000000015
pc ffffffc00000531c va/inst 0000000010000005 sr 8000000200006100
Kernel load segfault @ 0x0000000010000005
```

I also tested `pk` from `macOS`, then exited normally:

```
/usr/local/riscv/bin/spike ~/pk a.out
bbl loader
```

makutak, 2023-05-14:
I faced the same problem.
I just changed the commit hash(8ce2dc424eab98b907b3117b7515190c62f63c7f)  and re-built it and it worked.

My Distribution: Ubuntu 22.04

zacky1972, 2023-05-14:
Thank you, @makutak 

I can build by checking out the hash 8ce2dc4 and with `configure` option  `--with-arch=rv64gc_zifencei` according to the issue https://github.com/riscv-software-src/riscv-pk/issues/260

WojciechMula, 2023-05-15:
@makutak so, the commit https://github.com/riscv-software-src/riscv-pk/commit/54de960a5e4c91734fcfd454fdc7e593c6ac571d broke Ubuntu/Debian. Seems `is_32bit` is incorrectly set if we pass `--host` --- am I right?

WojciechMula, 2023-05-15:
@mga-sc could you please take a look?

mga-sc, 2023-05-16:
Yes, it seems like my commit https://github.com/riscv-software-src/riscv-pk/pull/296/files broke this. But my inner testsuite wasn't broken, I'm sorry.

So, in order to solve this issue, first we should roll back changes.

But the second problem is connected with csr and fence. We must include im_zicsr_zifencei in march. Can't we get info about rv32/rv64 from host if -march wasn't set?

aswaterman, 2023-05-16:
I reverted #296 until this is resolved.