riscv-software-src/riscv-pk #301: Can pk init vtype when specify V extension?

lhtin, 2023-08-11:
As this issue said (https://github.com/riscv-software-src/riscv-isa-sim/issues/1106), I think the pk should init vtype(`vsetivli x0, 0, e8, m1, ta, ma`) if V extension is enabled. Thanks.

lhtin, 2023-08-11:
For example, add this code at the bottom of minit.c file:
```c
   // Enable vector extension 
   if (supports_extension('V')) 
     mstatus |= MSTATUS_VS; 
  
   write_csr(mstatus, mstatus); 
 #if __riscv_xlen == 32 && __BYTE_ORDER__ == __ORDER_BIG_ENDIAN__ 
   write_csr(0x310, mstatush); /* mstatush is not known to gas */ 
 #endif 
  
   // Enable user/supervisor use of perf counters 
   if (supports_extension('S')) 
     write_csr(scounteren, -1); 
   if (supports_extension('U')) 
     write_csr(mcounteren, -1); 
  
   // Enable software interrupts 
   write_csr(mie, MIP_MSIP); 
  
   // Disable paging 
   if (supports_extension('S')) 
     write_csr(satp, 0); 
  
  // Init vtype, vlen csr
  if (supports_extension('V'))
    asm volatile ("vsetivli x0, 0, e8, m1, ta, ma");
```

aswaterman, 2023-08-11:
Only if an ABI requires this property, which AFAIK is not the case.

lhtin, 2023-08-11:
> Only if an ABI requires this property, which AFAIK is not the case.

Sorry for not understanding what you meant. Do you mean that the conditions for executing the vsetivli instruction are still uncertain inside the PK?

aswaterman, 2023-08-11:
I meant that pk should only be responsible for enforcing this constraint if the ABI also says that this constraint be employed upon entry to the program.

If the ABI says this state is unspecified upon entry to the program, then it isn’t pk’s responsibility to enforce it.

lhtin, 2023-08-11:
Oh, you mean the psABI specification[1] doesn't have a rule about that behavior, so PK isn't responsible for doing it? If the behavior is indeed necessary it needs to be added to the psABI specification first?

[1] https://github.com/riscv-non-isa/riscv-elf-psabi-doc

aswaterman, 2023-08-11:
I am speaking more generally than the psABI. In general, any ABI specifies what can be assumed about architectural state upon entry. If the ABI does not specify the state of this information on entry, then it is the program’s responsibility to initialize it. (Usually this happens in crt0 or some other early boot code; it isn’t the programmer’s responsibility.)

If the psABI is silent on the matter, then you have two options: change the userspace code, or request a change to the psABI. The latter might be slightly backwards incompatible, but if you do get it approved, we will update pk.

lhtin, 2023-08-11:
So is crt0 part of ABI?

aswaterman, 2023-08-11:
There are multiple ABIs in a system (e.g. kernel/low-level user vs. low-level user/application), and so that’s a question for people who don’t participate in this forum. Let us know when you figure out the answer and we’ll update pk if need be.

lhtin, 2023-08-11:
OK. Thank you so much for your detailed answer. I will close this issue currently and reopen it if needed.

closed 2023-08-11 by lhtin