riscv-software-src/riscv-pk #307: Error While configuring  Berkeley boot loader

sramichetty20019, 2023-11-10:
Hello Everyone,

I am trying to run linux kernel on rocketchip , i believe this is possible using the Berkeley boot loader so once i build the rocketchip i get an executable, which i can be used something like this ..
command:  ./emulatorDefaulrocketconfig bbl <path to my binary>

The reason to do this is my binaries have syscalls which cannot be used along with proxy kernel 

For this purpose, I am trying to configure the Berkeley bootloader 
 I generated the vmlinux kernel binary 
then I ran the following steps in riscv-pk directory 
mkdir build
cd build
 ./configure --prefix=$RISCV --host=riscv64-unknown-elf --with-payload=linux-4.6.2/vmlinux
make 

I encountered the following error :
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"linux-4.6.2/vmlinux\" -I. -I./bbl -c ./bbl/configstring.c
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"linux-4.6.2/vmlinux\" -I. -I./bbl -c ./bbl/kernel_elf.c
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"linux-4.6.2/vmlinux\" -I. -I./bbl -c ./bbl/logo.c
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"linux-4.6.2/vmlinux\" -I. -I./bbl -c ./bbl/mtrap.c
./bbl/mtrap.c: Assembler messages:
./bbl/mtrap.c:130: Error: unrecognized opcode `fence.i', extension `zifencei' required
make: *** [Makefile:300: mtrap.o] Error 1

If I execute the above steps in other way I get the following error :

Steps:
mkdir build 
cd build 
 sudo ../configure --host=riscv64-unknown-linux-gnu --with payload=/home/srinija/linux/vmlinux --prefix=/opt/riscv/ 
sudo make -j 24
Error :
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
make: *** [Makefile:336: file.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [Makefile:336: syscall.o] Error 1
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
make: *** [Makefile:336: handlers.o] Error 1
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
make: *** [Makefile:336: frontend.o] Error 1
make: *** [Makefile:336: console.o] Error 1
make: *** [Makefile:336: usermem.o] Error 1
make: *** [Makefile:336: elf.o] Error 1
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
make: *** [Makefile:336: mmap.o] Error 1
make: *** [Makefile:336: entry.o] Error 1

Can anyone please tell me how to solve the above errors, I also configured riscv-tool-chain properly, I have both newlib and glibc toolchain flavors installed

Regards
Srinija

Lester-1, 2023-12-11:
I also encountered this problem, did you find a solution?

sramichetty20019, 2023-12-14:
I couldn't fix it, however, I posted the same question on Chipyard Community this was the response I received. I hope it's useful for you ...

"BBL is no longer maintained or supported. Please use OpenSBI. 

A flow demonstrating how to boot Linux with OpenSBI is included in Firemarshal, which is part of Chipyard. "