riscv-software-src/riscv-pk #329: Issue while installing the riscv-pk -- ../pk/pk.c:188: Error: unknown CSR `senvcfg'

advaneharshal, 2024-06-20:
I am getting the following error while doing the installation of riscv-pk 

Steps I followed 
mkdir build 
cd build 
../configure --prefix=/home/advanehs/riscv_toolchain/riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-ubuntu14/ --host=riscv64-unknown-elf 
make 
At this step I am getting the following error 
../pk/pk.c: Assembler messages:
../pk/pk.c:188: Error: unknown CSR `senvcfg'
../pk/pk.c:188: Error: unknown CSR `senvcfg'

Please let me know how  to resolve the same 

gcc version 12.3.0 (Ubuntu 12.3.0-1ubuntu1~22.04)
Ubuntu 12.3.0-1ubuntu1~22.04

leandro-ro, 2024-06-28:
same for me, but with `riscv32-unknown-elf`

aswaterman, 2024-06-28:
You need a more recent assembler. `senvcfg` was added to binutils in early 2022, so probably version 2.39 or later would work.  (I'm using 2.41.)

As a temporary workaround, you can delete the problematic two lines of code:

```diff --git a/pk/pk.c b/pk/pk.c
index dbfaa5a..bb60f79 100644
--- a/pk/pk.c
+++ b/pk/pk.c
@@ -184,8 +184,6 @@ static void run_loaded_program(size_t argc, char** argv, uintptr_t kstack_top)
   init_tf(&tf, current.entry, stack_top);
   __riscv_flush_icache();
   write_csr(sscratch, kstack_top);
-  if (zicfilp_enabled)
-    set_csr(senvcfg, SENVCFG_LPE);
   start_user(&tf);
 }
 
```