riscv-software-src/riscv-pk #331: Issue when installing PK for RV32E

koni-ey, 2024-07-05:
I want to set up a simulation environment for a RV32EMFC architecture. 
(Spike apperently supports RV32E so i assumed PK does too).

Toolchain seems to be correctly installed as `riscv32-unknown-elf-gcc -Q --help=target` reports: 
```
-mabi = ilp32e
-march = rv32emfc_ziscr
```

I now tried to compile pk with `../configure --prefix=$RISCV --host=riscv32-unknown-elf --with-arch=rv32emfc_zicsr --with-abi=ilp32e` unfortunately I get the following output:
``` 
riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32emfc_zicsr -mabi=ilp32e -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/file.c
riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32emfc_zicsr -mabi=ilp32e -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/syscall.c
riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32emfc_zicsr -mabi=ilp32e -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/handlers.c
riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32emfc_zicsr -mabi=ilp32e -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/frontend.c
riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32emfc_zicsr -mabi=ilp32e -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/elf.c
riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32emfc_zicsr -mabi=ilp32e -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/console.c
riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32emfc_zicsr -mabi=ilp32e -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/mmap.c
riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32emfc_zicsr -mabi=ilp32e -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/usermem.c
riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32emfc_zicsr -mabi=ilp32e -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/entry.S
../pk/entry.S: Assembler messages:
../pk/entry.S:22: Error: illegal operands `sw x16,16*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:23: Error: illegal operands `sw x17,17*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:24: Error: illegal operands `sw x18,18*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:25: Error: illegal operands `sw x19,19*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:26: Error: illegal operands `sw x20,20*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:27: Error: illegal operands `sw x21,21*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:28: Error: illegal operands `sw x22,22*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:29: Error: illegal operands `sw x23,23*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:30: Error: illegal operands `sw x24,24*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:31: Error: illegal operands `sw x25,25*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:32: Error: illegal operands `sw x26,26*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:33: Error: illegal operands `sw x27,27*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:34: Error: illegal operands `sw x28,28*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:35: Error: illegal operands `sw x29,29*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:36: Error: illegal operands `sw x30,30*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:37: Error: illegal operands `sw x31,31*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:44: Error: illegal operands `csrr t3,scause'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:49: Error: illegal operands `sw t3,35*(1<<2)(x2)'
../pk/entry.S:61:  Info: macro invoked from here
../pk/entry.S:94: Error: illegal operands `lw x16,16*(1<<2)(a0)'
../pk/entry.S:95: Error: illegal operands `lw x17,17*(1<<2)(a0)'
../pk/entry.S:96: Error: illegal operands `lw x18,18*(1<<2)(a0)'
../pk/entry.S:97: Error: illegal operands `lw x19,19*(1<<2)(a0)'
../pk/entry.S:98: Error: illegal operands `lw x20,20*(1<<2)(a0)'
../pk/entry.S:99: Error: illegal operands `lw x21,21*(1<<2)(a0)'
../pk/entry.S:100: Error: illegal operands `lw x22,22*(1<<2)(a0)'
../pk/entry.S:101: Error: illegal operands `lw x23,23*(1<<2)(a0)'
../pk/entry.S:102: Error: illegal operands `lw x24,24*(1<<2)(a0)'
../pk/entry.S:103: Error: illegal operands `lw x25,25*(1<<2)(a0)'
../pk/entry.S:104: Error: illegal operands `lw x26,26*(1<<2)(a0)'
../pk/entry.S:105: Error: illegal operands `lw x27,27*(1<<2)(a0)'
../pk/entry.S:106: Error: illegal operands `lw x28,28*(1<<2)(a0)'
../pk/entry.S:107: Error: illegal operands `lw x29,29*(1<<2)(a0)'
../pk/entry.S:108: Error: illegal operands `lw x30,30*(1<<2)(a0)'
../pk/entry.S:109: Error: illegal operands `lw x31,31*(1<<2)(a0)'
make: *** [Makefile:336: entry.o] Error 1
```

It seems like its trying to acces registers > x15 which is weird with RV32E arch and abi specified. 

Did someone succeed building it for RV32E or set up a simulation environment for it?

hugoliu2019, 2024-12-20:
You can try 
../configure --prefix=$RISCV --host=riscv32-unknown-elf 
without other options

aswaterman, 2024-12-20:
PK doesn't support RV32E/RV64E, and we don't plan to add such support, since PK is meant to support the Linux ABI, which is based on RV32I/RV64I.