riscv-software-src/riscv-pk #339: Spike dose not recognize bset instruction

Ye-Jinhong, 2024-11-29:
When I am learning spike, I try to build a dhrystone to execute. However, I across a trouble flowing:
1. build gnu-toolchain by `./configure --prefix=$RISCV --with-arch=rv64gc_zba_zbb_zbc_zbs_zcb_zicond_zfa --with-abi=lp64d`
2. build spike by `../configure --prefix=$RISCV`
3. build pk by `../configure --prefix=$RISCV`

and dhrystone was compiled with `march=rv64gc_zba_zbb_zbc_zbs_zcb_zicond_zfa`. But when I run this:
```
spike pk dhrystone
```
`bset a0, zero, 0x21` is seen as a illegal instruction:
[image]

So I guest that shall I compile pk with `--with-arch`?
I rebuild pk by  
```
make clean
../configure --prefix=$RISCV --with-arch=rv64gc_zba_zbb_zbc_zbs_zcb_zicond_zfa
make
make install
```
**BUT** it even can't run a hello_world!
```
spike pk hello
```
It will stall for a long time, do nothing.

I dont know how to solve this problem. (T__T)

jerryz123, 2024-11-29:
You can set spike's isa with the --isa flag. Look at spike's help text.

Ye-Jinhong, 2024-11-29:
Thank you VERY MUCH!!!! The `--isa` should equal to pk's isa flag. Save me a lot of time!!! @jerryz123