riscv-software-src/riscv-pk #340: Debugging failures during elf loading

christian-herber-nxp, 2024-12-10:
I am currently facing a fail during loading of an elf file, and I have a hard time understanding what the issue could be.
In general, slightly more explicit error messages would be nice, but for that one has to understand the code.
My error is thrown on that condition in pk/elf.c:

```
size_t phdr_size = eh.e_phnum * sizeof(Elf_Phdr);
if (phdr_size > info->phdr_size)
```

I would appreciate if somebody has input as to what this indicates.

aswaterman, 2024-12-11:
There's a data structure that holds the program headers and is statically sized: https://github.com/riscv-software-src/riscv-pk/blob/abadfdc507d5a75b6272dc360e70a80a510c758a/pk/pk.c#L231

You could probably increase it without consequence.  But I agree the error message could be clearer in this case, since this error is not about a malformed ELF, but about a pk constraint.

christian-herber-nxp, 2024-12-21:
If the size of the header is anyway discovered dynamically, why is the memory to store it not allocated dynamically?
As the headers can be arbitrarily large, it seems difficult to hardcode this.

aswaterman, 2024-12-22:
I agree that would be better, but it is not a priority--you're the first person to have raised this issue.  If you submit a PR to that end, I'd be happy to review it.

christian-herber-nxp, 2024-12-22:
I will put it in the back burner.