riscv-software-src/riscv-pk #342: Misaligned double load and store are handled differently

dwRchyngqxs, 2025-03-27:
Misaligned double load and store are handled differently, the load is fine while the store triggers an unhandlable trap (and spikes exit with exit code 0).
The following main.S shows the issue (comment and uncomment the fld/fsd lines as you see fit).
```asm
.globl main
.text
main:
addi sp, sp, -9
fld ft0, 0(sp)
fsd ft0, 0(sp)
addi sp, sp, 9
mv a0, zero
jr ra
```
Output is
```
../machine/mtrap.c:22: machine mode: unhandlable trap 7 @ 0x80005b8c
Power off
```
when the misaligned store is encountered and nothing in all other cases.
Compiled with `riscv32-unknown-elf-gcc -o main -s main.S` and run with `spike --isa=rv32g pk main`.

Misaligned load and stores for integers complete successfully (no complaint at all, exit code 0).

Linked issue in spike in case it is a spike problem: https://github.com/riscv-software-src/riscv-isa-sim/issues/1943

aswaterman, 2025-03-27:
Thanks for the bug report.  I see what's going on; working on a fix.

aswaterman, 2025-03-29:
Turns out the issue was more subtle than I had guessed, but I believe it is fixed.

closed 2025-03-29 by aswaterman

dwRchyngqxs, 2025-03-29:
Thank you for fixing it this quick. I'm going to test if it is fixed and get back to you soon.

dwRchyngqxs, 2025-03-30:
I confirm the behaviour is now consistent.

commit d8659a4e (2025-03-29): Fix UB in RV32 versions of GET_F64_REG/SET_F64_REG

Resolves #342