riscv-software-src/riscv-pk #344: Getting error calling glibc's `arc4random`

thepalbi, 2025-04-29:
Hi everyone,

I'm trying to use `spike pk` to run a build of [OpenFHE](https://openfhe.org/) for iterating some optimizations I'm doing for RISC-V. I got almost everything building statically and else, but found the following error in one of their examples:

```
pablo@pablok:~/phd/openfhe-development/build_riscv$ spike pk bin/examples/pke/simple-real-numbers
Fatal glibc error: cannot get entropy for arc4random
```

Diving a bit deeper, found that `arc4random` is a random generator implemented in glibc, [as follows](https://sourceware.org/git/?p=glibc.git;a=blob;f=stdlib/arc4random.c;h=a250f3d9094709d75ae525ce1c5b92906136fd08;hb=HEAD). I did a minimal example as follows:

```c
#include <stdlib.h>
#include <iostream>

int main() {
  // Generate a random number using arc4random
  auto randomNumber = arc4random();

  // Print the random number
  std::cout << "Random number: " << randomNumber << std::endl;

  return 0;
}
```

And getting the same error. Reading the `arc4random` implementation, it seems it either opens `/dev/[?u]random` which I tried in a minimal example and it works, or it calls the `getrandom` syscall. Saw [this PR](https://github.com/riscv-software-src/riscv-pk/pull/300) that was closed some time ago. 

Any suggestions on working around this? I'm more than happy to open a PR and implement support for such a syscall.

thepalbi, 2025-04-29:
Tried doing a super simple `getrandom` syscall impl and it seems it's not that:
```
pablo@pablok:~/opt/riscv-pk$ git diff
diff --git a/pk/syscall.c b/pk/syscall.c
index dcd03fc..0e3b8bf 100644
--- a/pk/syscall.c
+++ b/pk/syscall.c
@@ -12,6 +12,8 @@
 #include <string.h>
 #include <errno.h>
 
+#include <fcntl.h>
+
 typedef long (*syscall_t)(long, long, long, long, long, long, long);
 
 struct iovec {
@@ -509,6 +511,14 @@ long sys_time(long* loc)
   return t;
 }
 
+ssize_t sys_getrandom(void *buf, size_t buflen, unsigned int flags)
+{
+  int urandom_fd = sys_open("/dev/urandom", O_RDONLY, 0);
+  ssize_t read_bytes = sys_read(urandom_fd, buf, buflen);
+  sys_close(urandom_fd);
+  return read_bytes;
+}
+
 int sys_times(long* loc)
 {
   uint64_t t = rdcycle64();
@@ -771,6 +781,7 @@ long do_syscall(long a0, long a1, long a2, long a3, long a4, long a5, unsigned l
     [SYS_readv] = sys_readv,
     [SYS_riscv_hwprobe] = sys_riscv_hwprobe,
     [SYS_futex] = sys_stub_success,
+    [SYS_getrandom] = sys_getrandom,
   };
 
   const static void* old_syscall_table[] = {
diff --git a/pk/syscall.h b/pk/syscall.h
index 979363a..77d536f 100644
--- a/pk/syscall.h
+++ b/pk/syscall.h
@@ -61,6 +61,8 @@
 #define SYS_readv 65
 #define SYS_riscv_hwprobe 258
 #define SYS_futex 98
+// printf SYS_getrandom | clang -include sys/syscall.h -E -
+#define SYS_getrandom 278
 
 #define OLD_SYSCALL_THRESHOLD 1024
 #define SYS_open 1024
```
Also, reading the code more carefully looks like upon receiving the `ENOSYS` from pk it [should default](https://sourceware.org/git/?p=glibc.git;a=blob;f=stdlib/arc4random.c;h=a250f3d9094709d75ae525ce1c5b92906136fd08;hb=HEAD#l55) to `/dev/urandom`, so I wonder if there's some problem elsewhere.

thepalbi, 2025-04-29:
Ok narrowed down to which part of the code is doing the call to `arc4random`, it's std::uniform_int_distribution [here](https://github.com/openfheorg/openfhe-development/blob/7b8346f4eac27121543e36c17237b919e03ec058/src/core/lib/utils/prng/blake2engine.cpp#L108).
```c
std::uniform_int_distribution<uint32_t> distribution(0);
```
Was able to bypass it with a development flag. Wonder if it's possible to implement support for it in pk (for `arc4random` I mean), and some advice on how to. Thanks!

Worst case I can use QEMU for emulating what I need on top of risc-v.

aswaterman, 2025-04-29:
Given typical uses of `pk`, we probably want something deterministic, even though that runs counter to the normal goals of using an RNG.  I'll make a PR that adds the syscall in that fashion.

aswaterman, 2025-04-30:
Done.  HTH.

commit 02b1e217 (2025-04-30): Add exceedingly fake version of sys_getrandom

Resolves #344
closed 2025-04-30 by aswaterman